Doodle3D/Doodle3D-cura-plugin #10: Handle all states

peteruithoven, 2015-12-18:
Make a lost of all possible states and make sure we give feedback when appropriate. 
- [ ] idle
- [ ] idle
- [ ] idle

peteruithoven, 2016-01-19:
Are you sure you've added feedback on that it's (re)slicing? 
Are you sure you're disabling the buttons when the connection to the WiFi-Box is lost?

mpbomil, 2016-01-19:
I just remembered that we are going to show feedback by using the slicing progress into our progress bar. We'll need [this](https://github.com/Ultimaker/Cura/blob/379ba357cf0b033cfe6ef626c613131fe2299616/plugins/CuraEngineBackend/CuraEngineBackend.py#L99 ) signal to detect the starting of a slicing. And [this](https://github.com/Ultimaker/Cura/blob/379ba357cf0b033cfe6ef626c613131fe2299616/plugins/CuraEngineBackend/CuraEngineBackend.py#L191) to give us a percentage between 0 and 100%. I'll look into it. I'll keep this issue open till I've added the signals.

As for the buttons, they are are auto-disabling when the box loses connection with the printer, but turning off the printer would not, because the latter can not get information from /d3dapi/info/status to check the state. I've added a timeout of 5 seconds in the "GET" function. So basically: if we have no answer in 5 seconds after requesting printer info, the plugin will assume the printer is off and it will set the printer state to "disconnected", the buttons will behave accordingly.

[Added timeout of 5 seconds @ PrinterConnection.py line. 441](https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L441)
[Setting printer state to disconnected @ PrinterConnection.py line. 357](https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L357)

mpbomil, 2016-01-19:
It seems there are two possible ways we can have the disconnect status.
1: When the Wi-Fi box is on but not connected to the printer physically.
2: When the Wi-Fi box was on, but now off. (Printer turned off/No power)

I have them both set to the printer state "disconnected", since there doesn't need to be a distinction (yet).

peteruithoven, 2016-01-19:
I would personally add a separate state, since the interface should show slightly different feedback. Maybe something like "unavailable" or "unknown"?
The benefit would be that you could create a updatePhrase() function that could set all possible phrases according to the different states.

mesut-pehlivan, 2016-02-24:
I seperated them into two states: "disconnected" for when the Doodle3D WiFi-Box is not connected to the printer, and "lost" for when the connection with the Doodle3D WiFi-Box is lost.

[lost L.354](https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L354)
[disconnected L.365](https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L365)