Doodle3D/Doodle3D-cura-plugin #13: When buffer full, delay before retry?

peteruithoven, 2016-01-08:
If I understand the `printGCode` logic there is no delay after it detects the buffer is full, this means it would retry this as fast as possible. Shouldn't we add a little delay here?

mpbomil, 2016-01-19:
[Added 3 seconds delay for keeping buffered lines in check. @ PrinterConnection.py line. 232](https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L232)

peteruithoven, 2016-01-19:
But now it always adds a 3 second delay between sending print parts?

mpbomil, 2016-01-19:
Couldn't I just delete the time.sleep here: https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L232
Since https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L227 does the job now?

peteruithoven, 2016-01-19:
You could, but I can imagine you want to use different timeout times for those.

mpbomil, 2016-01-19:
This should fix that:
https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L232

peteruithoven, 2016-01-19:
I think that makes sense.

closed 2016-01-19 by peteruithoven