Doodle3D/Doodle3D-cura-plugin #14: Optimistic state logic issue

peteruithoven, 2016-01-19:
There is logic added to override the state, to optimistically assume the printer will start printing for example. But the following line prevents the api from correcting the state when this optimistic assumption was wrong. 
https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L377

I would try to release the state before doing the http request
https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L351
This would allow the api to correct the state, after an override, when it starts a new request.

mpbomil, 2016-01-19:
There's another place where the statelock gets released:
https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L224

I've been thinking about putting the releasestatelock in a more logical place but haven't come up with the best place yet.

peteruithoven, 2016-01-19:
I would place it above: https://github.com/Doodle3D/Doodle3D-cura-plugin/blob/master/PrinterConnection.py#L351