Doodle3D/Doodle3D-cura-plugin #19: Integrate interface into main window?

peteruithoven, 2016-01-22:
Couldn't we integrate the interface into the main window?
See for example the lower right corner of the latest version of Cura. It even looks like there is a progressbar already... Very nicely styled in the style of the rest of the interface...
[image]
This would also solve: https://github.com/Doodle3D/Doodle3D-cura-plugin/issues/18

Remember how Gimp used to be with the separate windows?
http://cdn2.pcadvisor.co.uk/cmsdata/reviews/3280572/GIMP_496.jpg
Now all the windows are foldouts in the main interface, like Illustrator and Photoshop:
http://www.gimp.org/news/images/2015-report/2015-12-31-pfm-processing.jpg

peteruithoven, 2016-01-26:
Interesting issue to follow: https://github.com/Ultimaker/Cura/issues/596

> In the end we decided on dividing processes in two classes: processes that are blocking to the workflow (though not necessarily to the application), such as loading an object, and processes that happen in the background and allow the user to continue using Cura, such as slicing or printing. The former progress is shown in the Message Stack, the latter in the Sidebar.

mesut-pehlivan, 2016-02-24:
This was answered by Ultimaker:
> 2. Wie ontwikkelt op dit moment de USB printing plugin? Wellicht kan ik met
> diegene in contact komen over o.a. de designkeuzes en hebben we meer
> stroomlijning met onze plugins.

USB printing plugin wordt op dit moment door ons ontwikkeld, maar met minimale
effort aangezien het vooral een legacy support ding is. Jaime is degene hier
die hem oorspronkelijk heeft geschreven.

> Ik vraag me bijvoorbeeld af hoe de USB printing window die er nu is -
> verder ontwikkeld gaat worden.
> Zijn er plannen om de voortgang van de prints in de main window te doen, of
> is dat uitsluitend voor de slicing voortgang?

Ja, die zijn er wel, maar dat zal pas in een volgende versie er in komen. Voor
nu is dezelfde oplossing als voor de USB printing de beste.

closed 2016-02-24 by mesut-pehlivan